Galzai/MK32 #2: ESP32 does not show up in BLE scans

bilogic, 2018-12-30:
Hi,

I'm using the WeMos mini32 v1.0.0, it is just the ESP32 by itself with no switches wiring or OLEDs etc. Just wanted to get the software working before going onto the hardware.

MK32 managed to compile with 3 warnings and booted. However, it could not be found when scanning with Windows 10, Android 6.0.1 and the nRF android app. 

I'm expecting to find the default name of ```MK32 V2.1```, below are my ```make flash``` compiler messages, ```make monitor``` output and my ```keyboard_config.h```, any help is greatly appreciated. 

Thank you!

```
$ make flash
...
CC build/main/plugins/plugin_components/wifi_manager/wifi_manager.o
E:/Temp/esp/msys32/esp/MK32/plugins/plugin_components/wifi_manager/wifi_manager.c: In function 'wifi_connection_init':
E:/Temp/esp/msys32/esp/MK32/plugins/plugin_components/wifi_manager/wifi_manager.c:432:49: warning: passing argument 2 of 'esp_wifi_set_config' from incompatible pointer type [-Wincompatible-pointer-types]
       err = esp_wifi_set_config(ESP_IF_WIFI_STA,&ap_config);
                                                 ^
In file included from E:/Temp/esp/msys32/esp/MK32/plugins/plugin_components/wifi_manager/wifi_manager.h:13:0,
                 from E:/Temp/esp/msys32/esp/MK32/plugins/plugin_components/wifi_manager/wifi_manager.c:1:
E:/Temp/esp/esp-idf/components/esp32/include/esp_wifi.h:759:11: note: expected 'wifi_config_t * {aka union <anonymous> *}' but argument is of type 'wifi_sta_config_t * {aka struct <anonymous> *}'
 esp_err_t esp_wifi_set_config(wifi_interface_t interface, wifi_config_t *conf);
           ^
...
CXX build/main/components/nkolban_BLE/BLECharacteristic.o
E:/Temp/esp/msys32/esp/MK32/components/nkolban_BLE/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setValue(float&)':
E:/Temp/esp/msys32/esp/MK32/components/nkolban_BLE/BLECharacteristic.cpp:721:17: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  *((float *)temp) = data32;
                 ^
E:/Temp/esp/msys32/esp/MK32/components/nkolban_BLE/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setValue(double&)':
E:/Temp/esp/msys32/esp/MK32/components/nkolban_BLE/BLECharacteristic.cpp:727:18: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  *((double *)temp) = data64;
                  ^
...
CXX build/nkolban_BLE/BLECharacteristic.o
E:/Temp/esp/msys32/esp/MK32/components/nkolban_BLE/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setValue(float&)':
E:/Temp/esp/msys32/esp/MK32/components/nkolban_BLE/BLECharacteristic.cpp:721:17: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  *((float *)temp) = data32;
                 ^
E:/Temp/esp/msys32/esp/MK32/components/nkolban_BLE/BLECharacteristic.cpp: In member function 'void BLECharacteristic::setValue(double&)':
E:/Temp/esp/msys32/esp/MK32/components/nkolban_BLE/BLECharacteristic.cpp:727:18: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  *((double *)temp) = data64;
                  ^                  
```
```
$ make monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from E:/Temp/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on COM18 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5940
load:0x40078000,len:9712
load:0x40080400,len:6328
entry 0x40080728
I (28) boot: ESP-IDF v3.3-beta1-136-g97eecfa1b 2nd stage bootloader
I (28) boot: compile time 12:23:24
I (37) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
[... 271 lines cut ...]
#define COL2ROW
//#define ROW2COL

//Encoder definitions
// #define R_ENCODER // undefine if no rotary encoder is used
//#define R_ENCODER_SLAVE // undefine if no rotary encoder is used on slave pad
#define ENCODER_A_PIN GPIO_NUM_32 // encoder phase A pin
#define ENCODER_B_PIN GPIO_NUM_33 // encoder phase B pin
#define ENCODER_S_PIN GPIO_NUM_27 // encoder switch pin

//OLED Parameters
// #define OLED_ENABLE //undefine if no oled is used
#define OLED_SDA_PIN GPIO_NUM_23
#define OLED_SCL_PIN GPIO_NUM_22

/*Battery monitoring
 * Please read check battery_monitor.h for resistor values before applying
 * use ADC1 only,  */

// #define BATT_STAT				//define to enable battery monitoring
#define BATT_PIN ADC1_CHANNEL_7 //gpio pin 35, refer to the esp32 before modifying

//deep sleep parameters, mind that reconnecting after deep sleep might take a minute or two
#define SLEEP_MINS 5 // undefine if you do not need deep sleep, otherwise define number of minutes for deepsleep
```

bilogic, 2018-12-30:
To isolate the cause, I decided to test out ```esp32_mouse_keyboard``` and it worked out of the box, both Windows 10 and Android detected ```FLIPMOUSE```.

This rules out hardware and probably, my compiler environment settings.

Galzai, 2019-02-15:
Hi,
I apologize for the (extremely) late response, I haven't noticed the issue being raised for some reason.
Have you managed to get it running? 
If not, I suspect the issue could arise because esp-idf might have been updated since the last time I've updated this repository, thus requiring me to update the Bluetooth component.
I think I'll have time to look into in the following days.

bilogic, 2019-02-15:
Hi,

Better late than never. :)
No, I have not gotten it to work. Please share if you do find the cause, as your project is much nearer to my goal than ```esp32_mouse_keyboard```. Thank you.

mixashin, 2019-02-25:
Hey @Galzai, first thank you for sharing this awesome project much appreciated!
Did you get a chance to look at this issue? Can you maybe direct me to where to start looking i want to take a shot at fixing this if you dont have time?

Galzai, 2019-02-26:
> Hey @Galzai, first thank you for sharing this awesome project, much appreciated!
> Did you get a chance to look at this issue? Can you maybe direct me to where to start looking i want to take a shot at fixing this if you dont have time?

Hey, unfortunately I didn't have the time to look into it yet, but what I would basically do is compare the BLE components:
https://github.com/Galzai/MK32/tree/master/components/nkolban_BLE

to the corresponding components in:
https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils
 
In order to check what updates have been made to them (since again I assume esp-idf has been updated since I last had time to work on this). 
You could also try using an older version of esp-idf until I update the repository (I think it might've been version 3.0 but I'm not sure) if you do attempt to use an older version of esp-idf it would be great if you could reply which version works so I could temporarily update the readme until I fix the issues).
I might have time to try it out next week but unfortunately I'm not sure.

Galzai, 2019-04-14:
Hey, sorry it took me such a long time to get to it, but everything should work now.
I replaced the bluetooth components entirely, and it seems to work well on my device now, let me know if everything works so I could close the issue!

closed 2019-04-14 by Galzai

bilogic, 2019-04-14:
I managed to compile and pair with my phone. Thanks!

boykhocnhe, 2019-09-19:
i can pair with my ipad, android phone but my macbook do't show up :(