Galzai/MK32 #4: Sending mouse commands

bilogic, 2019-04-19:
Hi,

I copied the function ```uart_console```  from https://github.com/asterics/esp32_mouse_keyboard to try sending mouse commands from the serial monitor.

However, the keys "a" and "d" does nothing (i.e. does not move left/right) while "w" and "s" moves the cursor left and right ("w" is supposed to move up and "s" supposed to move down).

Any one can spot where I gone wrong? Thank you.

```
typedef struct mouse_command
{
    int8_t x;
    int8_t y;
    int8_t wheel;
    uint8_t buttons;
} mouse_command_t;

typedef enum
{
	PRESS,
	RELEASE,
	PRESS_RELEASE,
	RELEASE_ALL
} keyboard_action;
typedef struct keyboard_command
{
	/** @brief type of this keyboard action */
	keyboard_action type;
	/** list of keycodes+modfiers to be pressed/released.
   * @note Low byte contains the keycode, high byte any modifiers
   * */
	uint16_t keycode;
} keyboard_command_t;

#define MOUSE_SPEED 30
#define CONSOLE_UART_TAG "CONSOLE_UART"
// serial port of monitor and for debugging
#define CONSOLE_UART_NUM UART_NUM_0

void uart_console(void *pvParameters)
{
	char character;
	mouse_command_t mouseCmd;
	keyboard_command_t keyboardCmd;

	//Install UART driver, and get the queue.
	uart_driver_install(CONSOLE_UART_NUM, UART_FIFO_LEN * 2, UART_FIFO_LEN * 2, 0, NULL, 0);

	ESP_LOGI("UART", "console UART processing task started");

	while (1)
	{
		// read single byte
		uart_read_bytes(CONSOLE_UART_NUM, (uint8_t *)&character, 1, portMAX_DELAY);
		// uart_parse_command(character, &cmdBuffer);

		if (halBLEIsConnected() == 0)
		{
			ESP_LOGI(CONSOLE_UART_TAG, "Not connected, ignoring '%c'", character);
		}
		else
		{
			//Do not send anything if queues are uninitialized
			if (mouse_q == NULL || keyboard_q == NULL || joystick_q == NULL)
			{
				ESP_LOGE(CONSOLE_UART_TAG, "queues not initialized");
				continue;
			}
			switch (character)
			{
			case 'a':
				mouseCmd.x = -MOUSE_SPEED;
				mouseCmd.y = 0;
				mouseCmd.buttons = 0;
				mouseCmd.wheel = 0;
				xQueueSend(mouse_q, (void *)&mouseCmd, (TickType_t)0);
				ESP_LOGI(CONSOLE_UART_TAG, "mouse: a");
				break;
			case 's':
				mouseCmd.x = 0;
				mouseCmd.y = MOUSE_SPEED;
				mouseCmd.buttons = 0;
				mouseCmd.wheel = 0;
				xQueueSend(mouse_q, (void *)&mouseCmd, (TickType_t)0);
				ESP_LOGI(CONSOLE_UART_TAG, "mouse: s");
				break;
			case 'd':
				mouseCmd.x = MOUSE_SPEED;
				mouseCmd.y = 0;
				mouseCmd.buttons = 0;
				mouseCmd.wheel = 0;
				xQueueSend(mouse_q, (void *)&mouseCmd, (TickType_t)0);
				ESP_LOGI(CONSOLE_UART_TAG, "mouse: d");
				break;
			case 'w':
				mouseCmd.x = 0;
				mouseCmd.y = -MOUSE_SPEED;
				mouseCmd.buttons = 0;
				mouseCmd.wheel = 0;
				xQueueSend(mouse_q, (void *)&mouseCmd, (TickType_t)0);
				ESP_LOGI(CONSOLE_UART_TAG, "mouse: w");
				break;
			case 'l':
				mouseCmd.x = 0;
				mouseCmd.y = 0;
				mouseCmd.buttons = 1;
				mouseCmd.wheel = 0;
				xQueueSend(mouse_q, (void *)&mouseCmd, (TickType_t)0);
				mouseCmd.x = 0;
				mouseCmd.y = 0;
				mouseCmd.buttons = 0;
				mouseCmd.wheel = 0;
				xQueueSend(mouse_q, (void *)&mouseCmd, (TickType_t)0);
				ESP_LOGI(CONSOLE_UART_TAG, "mouse: l");
				break;
			case 'r':
				mouseCmd.x = 0;
				mouseCmd.y = 0;
				mouseCmd.buttons = 2;
				mouseCmd.wheel = 0;
				xQueueSend(mouse_q, (void *)&mouseCmd, (TickType_t)0);
				mouseCmd.x = 0;
				mouseCmd.y = 0;
				mouseCmd.buttons = 0;
				mouseCmd.wheel = 0;
				xQueueSend(mouse_q, (void *)&mouseCmd, (TickType_t)0);
				ESP_LOGI(CONSOLE_UART_TAG, "mouse: r");
				break;
			case 'q':
				ESP_LOGI(CONSOLE_UART_TAG, "received q: sending key y for test purposes");
				keyboardCmd.keycode = 28;
				keyboardCmd.type = PRESS_RELEASE;
				xQueueSend(keyboard_q, (void *)&keyboardCmd, (TickType_t)0);
				break;
			default:
				ESP_LOGI(CONSOLE_UART_TAG, "received: %d", character);
				break;
			}
		}
	}
}
```

Galzai, 2019-04-20:
The mouse_q in my implementation needs to receive a uint8_t array with length `HID_MOUSE_IN_RPT_LEN` and not  the mouse command struct.
Have a look at `r_encoder.c` where I implemented usage of mouse commands (I think the cases are pretty self explanatory) and modify your function accordingly:

https://github.com/Galzai/MK32/blob/4f9945987915d4af3e373efe4457ecd244d9567e/components/r_encoder/r_encoder.c#L68-L110

bilogic, 2019-04-20:
Thanks for info, I eventually got this to work with https://github.com/Galzai/MK32
```
typedef struct mouse_command
{
	uint8_t buttons;	// mouse_state[0]
	int8_t x;			// mouse_state[1]
	int8_t y;			// mouse_state[2]
	int8_t wheel;		// mouse_state[3]
} mouse_command_t;
```

The clue was from here, 
https://github.com/asterics/esp32_mouse_keyboard/blob/625440c04d0536783f51585b9161a9ae42c24686/components/nkolban_BLE/HID_kbdmousejoystick.cpp#L340-L358

It was confusing why the ```mouse_command_t``` had a different field order:
https://github.com/asterics/esp32_mouse_keyboard/blob/625440c04d0536783f51585b9161a9ae42c24686/components/nkolban_BLE/HID_kbdmousejoystick.h#L97-L104

closed 2019-04-20 by bilogic