Galzai/MK32 #6: Per key RGB backlight feature

bilogic, 2019-04-21:
Hi,

I was hoping to add per key RGB backlight.

The best effect would be to have dim lights as default and individual lights will go maximum brightness when their respective key is pressed and fade back to dim. Of course it should be flexible so that anyone else can add more effects as they wish.

The easiest approach would be to use the individually addressable WS2812B. They are much easier to wire, but consume more power.

Is plugin the way to go? (I tried to connect to the wifi manager plugin to check things out, but to no avail)

Thank you.

Galzai, 2019-04-21:
Unfortunately I didn't have time properly finish and test work on the plugin features (Including the wifi features), but I currently have more time so I'll resume working on it.

The idea for the plugin features though was more to create somewhat of a platform so people could easily write various plugins that could be enabled by assigning keys to them (so for example you could have a button that enables WiFi remote, Wifi server, Tetris on the oled, etc...)

Regarding back-light, I haven't really pursued that route yet, but you are more than welcome to add back-lighting  functionality and create a pull request!

bilogic, 2019-04-21:
Yes, I have already forked your project :)

But I'm still unfamiliar with the MK32 code base, I gather a plugin will need something like another ```xQueueSend``` inside the ```halBLETask_keyboard()``` and other ```halBLETask_*()```? The tricky part seems to be the modifier keys, e.g. Ctrl, Alt etc. Any tips/ideas?

Galzai, 2019-04-21:
Well, regarding back-lighting features, if you want them to correlate to keys being pressed, you can create a new task which basically just checks the MATRIX_STATE matrix (which is an array of rows that state which of the buttons are currently pressed, and is always updated).
I would go around do it to something like this:
in keyboard_config.h add:
`#define BACKLIGHT // uncomment if backlight is enabled`

in mk32_main.cpp:
```
#ifdef BACKLIGHT
	xTaskCreatePinnedToCore(backlight, "backlight task", 4096, NULL, configMAX_PRIORITIES, NULL,1);
	ESP_LOGI("Backlight","initializezd");
#endif
```
and than just write the backlight task function that should run in a loop and do something according to keys pressed:
```
extern "C" void backlight(void *pvParameters){

	for(uint8_t col = 0; col < MATRIX_COLS; col++){
		for(uint8_t row=0; row < MATRIX_ROWS; row++){
                     if (MATRIX_STATE[row][col] == 1){
                        ...do something
                     }
              }
         }
}
```

closed 2019-05-29 by Galzai

GilDev, 2020-05-10:
Any news on this @bilogic?

bilogic, 2020-05-11:
@GilDev I have given up on bluetooth keyboards, they disconnect intermittently.
I'm almost done creating a Logitech compatible one, see https://github.com/bilogic/logitech-unifying-device

GilDev, 2020-05-14:
Interesting, my experience with Bluetooth keyboards was nice (although it was mainly Apple and Logitech keyboards mostly on macOS and iOS), will look at your repo, thanks!