Galzai/MK32 #7: Wake from sleep

bilogic, 2019-04-21:
I have been getting the following on my serial monitor.
```
E (185630) KEY_REPORT: going to sleep!

13 is level 1
14 is level 1
15 is level 1
0 is level 0
4 is level 0
```

Currently using GPIO16/17 for my encoder's phase A/B, but rotating it does not wake the MK32.

1, Can the encoder wake it up?
2. Does 0, 4, 13, 14 and 15 means only these GPIOs can wake it up?

Thank you.

Galzai, 2019-04-21:
Wake up from deep sleep can only be used with keypresses, you can disable it all together though.
Have a look at the wiki: https://github.com/Galzai/MK32/wiki/Power-Management

bilogic, 2019-04-21:
Is the ```going to sleep!``` same as deep sleep?
How many levels of sleep are there? 

Funny, I connected to the encoder GPIO 13/14, but rotating does not wake it.
I will try again. Thanks.

Galzai, 2019-04-21:
Yes, "going to sleep"  is just the message it prints out, there is only one type of sleep enabled on the device.
The encoder rotation might not be able to wake up from sleep (as far as I know). the encoder's button should be able to wake up from sleep (if your encoder has a button).

Galzai, 2019-04-28:
Closed because question has been answered.

closed 2019-04-28 by Galzai