Galzai/MK32 #8: get error at NVS storage

boykhocnhe, 2019-09-19:
i'm use IDF sdk v4.0 to build the program
it work immediately so i can connect and typing as normal keyboard but i get error in
E (1129) NVS Storage: Error getting layout num: ESP_ERR_NVS_NOT_FOUND
E (1144) NVS Storage: Error getting layout name: ESP_ERR_NVS_NOT_FOUND
and when i run LAYOUT plugins it stop working

Galzai, 2019-09-19:
Hey,

That error should be fine (if no layouts are on NVS it reads the default layouts).
The LAYOUT plugin currently only opens a basic server that displays a message (I haven't had time to work on this project for a while).
When it stops working does it crash or does it just stop sending out keystrokes?

boykhocnhe, 2019-09-19:
i attached my debug log here
have add SSID and password in plugins.h
when i run LAYOUT plugins it scan available AP and it crash

Galzai, 2019-09-19:
Are you sure you sure you added the correct SSID and password? it seems when searching for AP's it does not find a match to the default one set.
Either way I think that for some reason if a connection fails I need to disable the event loop because it seems to cause the crash,I need to look into it.

boykhocnhe, 2019-09-19:
problem solved 
it my router problem
i create new wifi AP with my phone and it work perfectly

closed 2019-09-19 by boykhocnhe