Galzai/MK32 #13: it seems like that the keys near the row0 can not NKRO

shownb, 2020-05-29:
I tested a keyboard with project MK32
only one pad with col 3 row 5. connect to Mac OS.

| _ | col0 | col1 | col2 |
| ------------- | ------------- |------------- |------------- |
| row0  | 1  | 2  | 3  |
| row1  | q  | w  | e  |
| row2  | a  | s  | d  |
| row3  | z  | x  | c  |
| row4  | l  | m  | n  |

it is impossible to press down 1 and 2 together at the same time. the same situation like "1 and 3" "2 and 3".

but it can press down 6 keys like z x c l m n at the same time. but more than 6 keys it is also impossible.

Galzai, 2020-05-30:
I assume you have NKRO uncommented  in the keyboard_config file?
If so i'll try and look into it when I have more time,  ( last time i tested NKRO on my boards it worked as expected).

shownb, 2020-05-31:
col2row and there are no problems with the wire.
and I will try to find out what happen. 
you did a great job.thank you for sharing the code.you can close this issue.

closed 2020-05-31 by shownb