Galzai/MK32 #14: Macros not working properly?

cmpulsip, 2020-06-25:
I've followed the project for quite some time, and have recently built and started using my keyboard utilizing the esp32's. It works pretty much flawlessly, so I commend you on all the work you've done, but I have an issue with the macros. I believe I've defined them properly, but they're not working. I am using Macosx Catalina, I don't know if the operating system has something to do with this issue? I was wondering if you've been able to use them in a split keyboard application? I'm not sure what supporting documentation you need. Let me know and I'll do my best to get them to you if you're willing/able to address this issue.

Galzai, 2020-06-25:
Hey,
Just to make sure everything is set-up correctly, it would help to know:
* what actually happens when you set a macro and try to launch it.
* have you tried any of the pre-set macros, do they work?
* can you share your macro implementation? ( i.e. the changes you made to enable the macros).

cmpulsip, 2020-06-25:
1. From what I can tell, there isn't any visible response when I try to use a macro.
2. Macosx doesn't have an equivalent ctl+alt+delete, and that's the only macro that I see has been defined?
3. I have attached the keymap.c.
[keymap.c.zip](https://github.com/Galzai/MK32/files/4832063/keymap.c.zip)

Thanks for the quick reply. Looking forward to getting this working. Again, great work! Much appreciated!

Galzai, 2020-06-27:
I'll try and looks into it, but it will take some time, if you find the issue before I manage to get around to it it would be great if you could do a pull request!

huylesmartdevices, 2020-07-02:
@cmpulsip 
I think the issue due to the CMD key is not working on macOS. You can try with single key as KC_LGUI or KC_LCMD.

cmpulsip, 2021-02-22:
Hey, I know it's been a ridiculously long time since I raised this issue, but I finally got around to do some testing.

It looks like there may be an issue with the way the macros and modifiers are being populated in current_report. I changed the macro logic to be:

`uint16_t key = macros[keycode - MACRO_BASE_VAL][i];`

This change made some of the macros work, but then the modifier keys stopped working, such as shift, command, alt, etc...

Still looking into it, but if you had any ideas, I'd be open to any suggestions... Still poking around.