Galzai/MK32 #17: Can I use Pin34-39 as keyboard pins?

Jasonbroker, 2020-10-20:
I have a keyboard which has a 8*11 matrix. Because lolin32 lite only contains 19 output pin, there will be no other pins if I want to add a oled to it.
As the title said, could I use the 34,35,vp, vn pin for keyboard pins?

Galzai, 2020-10-21:
Technically you should be able to, only if you are using them as input pins (for example if you are using COL2ROW, they need to be used for rows).
You also might need to change the call to gpio_set_direction in matrix.c to use GPIO_MODE_INPUT  instead of GPIO_MODE_INPUT_OUTPUT (for rows), note that in my boards when I tried doing that for some reason those pins weren't set to 0 by default so I had to set to GPIO_MODE_INPUT_OUTPUT and manually change the values to 0, but it might be a specific issue with my board .

Jasonbroker, 2020-10-22:
thanks for replay.
I have tested that and the output is wired. technically this should work. I will look into it on weekend

closed 2020-10-30 by Galzai