Galzai/MK32 #20: error compiling with mingw32

jtpbryant, 2020-12-19:
im agetting this error when trying to make flash. can anyone help me interpret it.

AR build/battery_monitor/libbattery_monitor.a
/home/payto/esp/MK32/components/ble_new/component.mk:5: warning: undefined variable 'GCC_NOT_5_2_0'
CC build/ble_new/hal_ble.o
CC build/ble_new/hid_dev.o
CC build/ble_new/hid_device_le_prf.o
C:/msys32/home/payto/esp/MK32/components/ble_new/hid_device_le_prf.c:275:22: error: 'char_prop_read_write_notify' defined but not used [-Werror=unused-const-variable=]
 static const uint8_t char_prop_read_write_notify = ESP_GATT_CHAR_PROP_BIT_READ
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/msys32/home/payto/esp/MK32/components/ble_new/hid_device_le_prf.c:268:22: error: 'char_prop_notify' defined but not used [-Werror=unused-const-variable=]
 static const uint8_t char_prop_notify = ESP_GATT_CHAR_PROP_BIT_NOTIFY;
                      ^~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/payto/esp/esp-idf/make/component_wrapper.mk:292: hid_device_le_prf.o] Error 1
make: *** [C:\msys32\home\payto\esp\esp-idf/make/project.mk:635: component-ble_new-build] Error 2

ive been trying to figure out how to get this to work for 2 weeks now and i cant figure it out

Bupher, 2021-05-18:
I have the same issue, and also do not know how to correct it. Please let me know if you solved it.

Bupher, 2021-05-18:
Make sure you use ESP-IDF 4(.0.2)
I believe that is the problem

closed 2021-05-28 by Galzai