Galzai/MK32 #23: Battery monitor value

gijsn, 2021-03-14:
Hi,
Great project! Really comes in very useful for my single keyboard on the ESP32 (the sick68, no oled screen), I'll write up the wiki of how I built the thing once its done. Love the BLE functionality.

Though I have one minor issue, I cannot get the battery level to change on Windows. By default its set to 50% in the code, but all my attempts at modifying the attribute value have failed thus far. I saw the HID code is very similar to the esp-idf example, but unfortunately they also did not implement it. Could you give me some pointers to how I can adapt the value?

Let me know!
Gijs

gijsn, 2021-03-18:
I was finally able to achieve this: https://github.com/Galzai/MK32/compare/master...gijsn:master#diff-d0d45ac689cb047a071964d07f04cad0b774b3ccb7aeae83a0af1231368893f0R177

Not really sure if this is the proper way, but it works for me

Galzai, 2021-03-19:
If everything works for you feel free to make a pull request for this feature!

gijsn, 2021-03-19:
No worries, Ill cherry pick the code as I made some other changes as well and submit a PR

closed 2021-03-19 by gijsn