Galzai/MK32 #39: Docker build.sh fails

t3st-account, 2022-03-09:
running the build.sh script fails with the following message...

HOST: Linux mint uma 
─ uname -a                                                                 
Linux x270lm 5.13.0-30-generic #33~20.04.1-Ubuntu SMP Mon Feb 7 14:25:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

Sending build context to Docker daemon  36.74MB
Step 1/28 : FROM espressif/idf:release-v4.4
 ---> 29bbefec4ec6
Step 2/28 : RUN mkdir /home/workspace
 ---> Using cache
 ---> 4013c585bf54
Step 3/28 : RUN mkdir /home/workspace/cmakebuild
 ---> Using cache
 ---> d1cf621591e4
Step 4/28 : COPY components /home/workspace/components
 ---> Using cache
 ---> 41832d923c2b
Step 5/28 : COPY main /home/workspace/main
 ---> Using cache
 ---> ad15b3c91588
Step 6/28 : COPY CMakeLists.txt /home/workspace/
 ---> Using cache
 ---> 9fb08ec74906
Step 7/28 : COPY partitions.csv /home/workspace/
 ---> Using cache
 ---> 4bf32bbc1c50
Step 8/28 : COPY README.md /home/workspace/
 ---> Using cache
 ---> 70a5e0bffe2e
Step 9/28 : COPY sdkconfig.defaults /home/workspace/
 ---> Using cache
 ---> 772a4cbfa3e4
Step 10/28 : WORKDIR /home/workspace/cmakebuild
 ---> Using cache
 ---> 298dad072713
Step 11/28 : RUN /bin/bash -c "source /opt/esp/idf/export.sh"
 ---> Using cache
 ---> 1088e6301db1
Step 12/28 : ENV PATH="/opt/esp/tools/cmake/3.20.3/bin:${PATH}"
 ---> Using cache
 ---> ef2fea23d3fc
Step 13/28 : ENV PATH="/opt/esp/idf/components/esptool_py/esptool:${PATH}"
 ---> Using cache
 ---> 837c0dea1332
Step 14/28 : ENV PATH="/opt/esp/idf/components/espcoredump:${PATH}"
 ---> Using cache
 ---> 548200843c98
Step 15/28 : ENV PATH="/opt/esp/idf/components/partition_table:${PATH}"
 ---> Using cache
 ---> 20d3bebe6167
Step 16/28 : ENV PATH="/opt/esp/idf/components/app_update:${PATH}"
 ---> Using cache
 ---> 855b4a6fad71
Step 17/28 : ENV PATH="/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:${PATH}"
 ---> Using cache
 ---> e1aad8608578
Step 18/28 : ENV PATH="/opt/esp/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin:${PATH}"
 ---> Using cache
 ---> 2ca3b60452cc
Step 19/28 : ENV PATH="/opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin:${PATH}"
 ---> Using cache
 ---> 5ba63bb72126
Step 20/28 : ENV PATH="/opt/esp/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin:${PATH}"
 ---> Using cache
 ---> 9419a28b528f
Step 21/28 : ENV PATH="/opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:${PATH}"
 ---> Using cache
 ---> 9d6e3b93c224
Step 22/28 : ENV PATH="/opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:${PATH}"
 ---> Using cache
 ---> ac71111f63ef
Step 23/28 : ENV PATH="/opt/esp/tools/cmake/3.20.3/bin:${PATH}"
 ---> Using cache
 ---> a3efc2c6e870
Step 24/28 : ENV PATH="/opt/esp/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin:${PATH}"
 ---> Using cache
 ---> e02949556b55
Step 25/28 : ENV PATH="/opt/esp/python_env/idf4.4_py3.8_env/bin:${PATH}"
 ---> Using cache
 ---> a20295e7af6e
Step 26/28 : ENV PATH="/opt/esp/idf/tools:${PATH}"
 ---> Using cache
 ---> 9eaef8521d31
Step 27/28 : RUN cmake /home/workspace/
 ---> Running in d7a11de7a0da
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at /opt/esp/idf/tools/cmake/project.cmake:290 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

CMake Error at /opt/esp/idf/tools/cmake/project.cmake:290 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

CMake Error at /opt/esp/idf/tools/cmake/project.cmake:290 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/workspace/cmakebuild/CMakeFiles/CMakeOutput.log".
See also "/home/workspace/cmakebuild/CMakeFiles/CMakeError.log".

Galzai, 2022-03-12:
@dungphan90

Powerfred, 2022-04-09:
Same issue with docker sadly, is it possible to go the legacy route somehow?

Galzai, 2022-04-10:
> Same issue with docker sadly, is it possible to go the legacy route somehow?

You can try and follow the wiki:
https://github.com/Galzai/MK32/wiki/Building-and-flashing

alikthename, 2022-04-11:
also tried to compile docker build fails
Tried following instructions from https://github.com/Galzai/MK32/wiki/Building-and-flashing

MK32$ make flash ESPPORT=[/dev/ttyUSB0]
make: *** No rule to make target 'flash'.  Stop.

Is it possible to compile this thing now somehow?

Galzai, 2022-04-12:
Something might have changed on newer ESP-IDF you can try following the build guide:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-the-build-system

If this works for you, let me know the steps and I'll update the wiki or make a PR (I'm not actively working on this project for quite a while)

Powerfred, 2022-04-18:
I installed the ESP-IDF 4.4 for Windows 10 via https://dl.espressif.com/dl/esp-idf/?idf=4.4 
From the PowerShell Environment following 
```
git clone --recursive https://github.com/Galzai/MK32
cd .\MK32\
```
I could successfully flash the (so far unmodified) default build to a WROOM-32 dev board using
`idf.py -p COM4 clean flash`

(I am able to connect via BT from windows or android, but upon initiating a keypress it will get "stuck" and not release the key, no idea yet what that is about, but I very much assume it's unrelated to the flashing process)

closed 2023-02-26 by Galzai