Galzai/MK32 #40: What do you think about starting a community on Discord ?

cuminhbecube, 2022-04-16:
I have been interested in your project for quite some time. maybe 1 more than a year ago, but basically, this project is quite few people know or can know but can't contact anyone in the problems encountered by the project. In addition, recently espressif has released a number of new chip lines such as C3 or S3, what do you think about the project development for the above chip lines, and the Discord community's suggestions for further development of the project.
Maybe I don't speak English well, but here's what I can say about the project. Thanks

Galzai, 2022-04-18:
Feel free to open one, I can link up to it in the readme.
Note that this repo needs new maintainers, I haven't messed around with the ESP32 in quite a while (and frankly this project could be done a lot better< I started it without any experience)

xTye, 2022-05-13:
If this is still a thing, let me know, I'd be interested in developing

closed 2023-02-26 by Galzai