Galzai/MK32 #43: Many buttons doesn't work on android

vipaoL, 2022-08-05:
I tested the keyboard with my linux PC and it works perfectly but on android some buttons just don't do anything: not even the screen wakes up, although LED on my ESP blinks when i press the button, which doesn't work. This problem shows up with all my android devices

vipaoL, 2022-08-05:
[image]

vipaoL, 2022-08-05:
I also found that it doesn't work at all with postmarketOS (galaxy s3) although pairing and connecting work

Galzai, 2022-08-05:
If I remember correctly you can undefine the `NKRO `Macro in `keyboard_config.h` to get it to work on Android as well.

closed 2022-08-05 by vipaoL