Galzai/MK32 #45: Not working when used more than 12 column GPIO pins

weijieyong, 2022-10-22:
Hi, I'm trying to build a 5x15 (row x col) keyboard

I've edited
-  `#define MATRIX_COLS ` in `keyboard_config.h`
- `const gpio_num_t MATRIX_COLS_PINS[]` in `matrix.c`
- `uint16_t _QWERTY[MATRIX_ROWS][KEYMAP_COLS]` in `keymap.c`
but the keyboard is not working as expected after flashing.

It works fine when the number of columns is less than or equal to 12.
Am I missing something here?

weijieyong, 2022-10-22:
I tried logging it from `scan_matrix()` in `matrix.c` it was able to detect key presses,
but the character doesn't appear in my text field.

Galzai, 2022-10-23:
Might be an NKRO thing,  you can try to undefine the NKRO Macro in keyboard_config.h and see if it works-> Note that that will mean you only have 18 key rollover (which is fine for mostly everyone probably)

weijieyong, 2022-10-23:
Thank you for replying. I tried undefining NKRO Macro in keyboard_config.h but the problem still occurs.
I know you are not actively maintaining this, but any tips on figuring out the main cause of the problem would help. Thanks!

closed 2022-11-26 by weijieyong